sleemanj/optiboot #3: Java error when uploading or burning bootloader

MCUdude, 2017-10-21:
First of all, you have a lot of great work here! IMO it's a shame that it's so hard to find it (I was able to find it because you tagged me in [this post](https://github.com/SpenceKonde/ATTinyCore/issues/105)).

I was curious to test some of the ATtiny13 functionality you've added that MicroCore or Core13 doesn't have, but unfortunately I'm just getting av error every time I try to upload or burn bootloader:

```
"/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.0-arduino/bin/avr-objcopy" -O ihex -R .eeprom  "/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_611600/sketch_oct08a.ino.elf" "/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_611600/sketch_oct08a.ino.hex"
Sketch uses 568 bytes (55%) of program storage space. Maximum is 1024 bytes.
Global variables use 20 bytes (31%) of dynamic memory, leaving 44 bytes for local variables. Maximum is 64 bytes.
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:79)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:709)
	at processing.app.SketchController.exportApplet(SketchController.java:682)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2190)
	at java.lang.Thread.run(Thread.java:745)
```

Any idea how to get rid of this java error?

sleemanj, 2017-10-21:
Just going by the fact the error is somewhere in "SerialUploader", you're not trying to upload to a tiny13 over serial?  There is no room for a bootloader on the 13 so you need to use a programmer.  But then again the IDE shouldn't even try uploading over serial anyway for the 13 and default to programmer when you hit the button.

Otherwise it sounds like an IDE bug to me more than anything, a quick google seems to have similar conclusions.  Have you tried a different IDE version?

If you let me know what version IDE you're using I can give it a try myself.

MCUdude, 2017-10-21:
I get the same error with IDE 1.8.5 as well. I'm using MacOS. I don't even have a serial adapter connected, and I get the same error when shift+uploading.

sleemanj, 2017-10-21:
Hi, I'm using 1.8.5 on my Linux box and not seeing the issue.

Attached is an image showing the settings I have selected

[image]

Do you have any port available at all in Tools > Port? 

My first guess without doing any research would be that if you don't then maybe the IDE generates this error even if it's not going to use a serial port.  On my Linux box I have at least one serial port always so that may be something I wouldn't see myself.

If you do have a port listed, try selecting it even if it's not selected,  maybe even if you have a port listed there the IDE has become confused about it (maybe if you had one selected that has gone away for example).

Try plugging in a USB-Serial converter (or some other Arduino) and choosing it in Tools > Port to see if the error goes away,

MCUdude, 2017-10-21:
I always have two bluetooth ports available in the *Port* menu; they're there by default. I get the same error if I connect a USB to serial adapter and selects it. I assume there is something wrong in the platform.txt file that's causes this to fail on MacOS.

I ended up compiling the code in Arduino IDE and upload the hex file externally using avrdude in a terminal window.

[image]

sleemanj, 2017-10-21:
Which Programmer is selected (Tools > Programmer)?

MCUdude, 2017-10-21:
I'm using the default *USBtinyISP* option.

sleemanj, 2017-10-21:
Thanks, I don't have one of those (I use USBasp devices) but it still looks like it would work ok on my linux machine - at least it calls avrdude with the correct options.

I'll try and dig up a functional mac system and see if I can recreate the issue there.

sleemanj, 2017-10-21:
One more question, do you get the same error if you try say Board > ATTiny84, just to see if it's something specific to the Tiny13 boards.txt setup.

MCUdude, 2017-10-21:
I get the same error when selecting the USBasp too.
It's weird; I have about 10 different Arduino core installed, but never experienced this before..

MCUdude, 2017-10-21:
I get the same error when the ATtiny84 is selected.

MCUdude, 2017-10-21:
I think I found the issue!

```
attiny13_diy.upload.tool=diy_attiny:avrdude_noefuse
```
Needs to ba changed to:
```
attiny13_diy.upload.tool=avrdude
```

sleemanj, 2017-10-21:
Thanks :+1: , I'll check that out and roll it into the next release.

sleemanj, 2017-10-22:
FWIW, I wasn't able to reproduce in Mavericks with 1.8.5 :-/ 

[image]

I'll probably make the change you indicated anyway as I don't think it should make any difference on the 13 since there's no "upload" ability anyway without a bootloader.

Only other thing I could think of would be maybe if your "Arduino AVR Boards" board definitions stuck on an older version (Tools > Boards > Board Manager, first one in the list, latest version is 1.6.20).

per1234, 2018-02-09:
The syntax `diy_attiny:avrdude_noefuse` means a reference to the `diy_attiny` vendor's `avrdude_noefuse` tool. Vendor is set by the top level folder name of the package. I've never seen a package reference itself but it works fine for me whether I do a Boards Manager or a manual installation. However, there is no need to have the self-reference. This will work just as well:
```
attiny13_diy.upload.tool=avrdude_noefuse
```

The only cause of this problem I can think of is if MCUdude changed the vendor folder name from diy_attiny. When I do that I do get the same error message. If the unnecessary self-reference syntax is removed then the error won't occur regardless of the vendor folder name.

commit b0a6e323 (2022-03-12): Remove explicit reference to diy_attiny in tools.  Closes #3
closed 2022-03-12 by sleemanj