sleemanj/optiboot #8: Option for External Clock

wholder, 2018-04-14:
I have an application where I need to use an external TCXO for precision, so it would be nice to have the option to select an external clock.

sleemanj, 2018-08-28:
I've given this some thought, I don't think I will add general support for external clock because this could easily be selected by mistake by new people not understanding the difference between clock and crystal, if they pick a clock and they don't have a clock signal, they are boned until they sort out a suitable clock.

Also because of the structure of the menus I would have to select some combinations of TCXO and Frequency, and I don't know what people would want there, where as system crystal selection is typically "8, 16 or 20" for most users.

I don't think splitting the "Processor Speed" menu into "Clock Type" and "Frequency" would be a good idea either because obviously not all frequencies work on all clock types.

Maybe adding a 19.2 MHz TCXO Clock option would be OK since that's _fairly_ common and would not likely be selected accidentally.

Ultimately, somebody using a TCXO probably knows enough that they could burn the Xtal bootloader of the appropriate frequency, and then just use avrdude to change the fuses so that it is configured for the clock instead of xtal.

wholder, 2018-08-29:
First, thanks for taking the time to consider my suggestion and I understand the reasoning behind your decision not to support it.  But, I feel the need to persist and perhaps convince you to consider giving this some additional thought in the hope that you can come up with some alternative to my suggestion.

Perhaps the the boards menu is not the right place to handle this as, as you suggest, it might confuse novice users.  But the ability to select alternate options for the system clock is an important missing feature from the Arduino.  And, unfortunately, the design of the AVR series, only makes this possible by altering the fuses which are typically set when the boot loader is programmed.  So, i'm not sure what options this leaves for finding a clean way of being able to publish Arduino projects that require the use of either an external oscillator, or a non standard clock frequency. 

I have found other ways to work around the frequency issue, such as calculating and setting a non standard baud rate for the serial port, or adjusted timing values for the delay functions.  These are clumsy to use and potentially confusing for anyone reading the source code, but they are a work around (at least as long as I'm using ICSP to program the chip.)  But, I've yet to find a work around for specifying between an external clock or an external crystal and being able to do this is highly useful when using the ATTiiny series as it's a way to free up an additional I/O pin as well as use a TXCO.

Wayne

sleemanj, 2018-08-29:
> Perhaps the the boards menu is not the right place to handle this as, as you suggest, it might confuse novice users.

Unfortunately that's the only way that such options can be selected.  So as I say the best that would even be possible really is either

  1. Add a specific combination of frequency clock type to the "Boards > Processor Speed" menu (eg, "19.2MHz TXCO Clock Input" - the problem there is deciding what people might want to choose without flooding the menu with possible choices; or
  2. Split "Processor Speed" in to two independant menus, one "Clock Type" and one "Frequency" - the problem there is both that it increases the choices that the user has to make, and that the user can choose incompatible combinations (and we have no way to know or prevent that)

in other words, the Arduino IDE just doesn't allow for this sort of thing easily.

sleemanj, 2018-08-29:
What is the frequency of the TCXO you are using?

wholder, 2018-08-29:
My current project uses a 19.2 mHz crystal, but I actually have several
projects in the works that require nonstandard clock frequencies to
generate precise timing.  Another common frequency that would be useful is
15.36 mHz.

Wayne

sleemanj, 2018-09-09:
The release today, 2018.9.9 introduces "Override" menus, this allows you to select (override!) the Clock Source, Frequency and Bootloader Upload Speed.

Usage is to select the overrides you want, I recommend to select both clock source and frequency, leave upload speed as default, then be sure to burn bootloader in order to set the fuses.

This should allow you to configure at least the clocking as you wish.  Only very lightly tested, so please report any bugs with the fuse calculations when using Override menus.

Note that currently the menus are not sorted in a friendly way, the next Arduino IDE release should resolve that ( https://github.com/arduino/Arduino/issues/7955 )

closed 2018-09-09 by sleemanj

gwideman, 2022-01-24:
I would like to thank you for adding external clock capability. It's useful when working with raw QFN ICs in a socket, as the crystal oscillator may misbehave due to extraneous capacitance. (I do realize that for use on a PCB with a crystal the fuses will need to be changed for that.)