sleemanj/optiboot #18: add 0.6 MHz clock frequency

Master811129, 2020-03-16:
hello.
could you please add 0.6 MHz internal clock frequency. it is available if you set fuses to 4.8 MHz and then write a zero to CKDIV8.
it is super useful if you wanna a low power consumption MCU.
cheers.

Master811129, 2020-03-21:
like this one
[image]
600kHz clock frequency

sleemanj, 2020-03-22:
You can select 600kHz in the Override Clock Source and Override Frequency menu options.  Clock source alters the fuse setting, frequency alters the compiled frequency, set them both 600kHz, burn bootloader to set the fuses, and go from there.

Master811129, 2020-03-22:
> You can select 600kHz in the Override Clock Source and Override Frequency menu options. Clock source alters the fuse setting, frequency alters the compiled frequency, set them both 600kHz, burn bootloader to set the fuses, and go from there.

thanks for help James. but why didn't you add this clock frequency into"processor speed" menu and created another menu?

closed 2020-03-22 by Master811129

sleemanj, 2020-03-22:
Operation below 1MHz is not recommended, timing (millis, delay etc) may not operate as expected.  So I regard it as "for experts" only.

---- On Mon, 23 Mar 2020 03:37:33 +1300 notifications@github.com wrote ----

You can select 600kHz in the Override Clock Source and Override Frequency menu options. Clock source alters the fuse setting, frequency alters the compiled frequency, set them both 600kHz, burn bootloader to set the fuses, and go from there.

thanks for help James. but why didn't you add this clock frequency into"processor speed" menu and created another menu?