sleemanj/optiboot #21: avrdude: AVR Part "attiny13a" not found.

caiopoit, 2020-12-02:
Trying to upload a sketch gives this error for the attiny13a, selecting the attiny13 gives invalid device signature...
Strange thing is I didn't use to have this issue before... I tested arduino IDE 1.8.2 and 1.8.13. Both with boards updated and the older version of this CORE.

Archiving built core (caching) in: C:\Users\KylePolt\AppData\Local\arduino_cache_338607\core\core_diy_attiny_avr_attiny13_diy_version_attiny13a,boot_none,clock_4_8MHz,core_MillisCore,millis_ppm_1p6,print_BinHexDec,serial_HalfDuplexRW,lto_enable,bodlevel_2v7,advclk_auto,advfrq_auto_42fabf9675cd091f74dcb7bc3fd3cb53.a
Sketch uses 444 bytes (43%) of program storage space. Maximum is 1024 bytes.
Global variables use 10 bytes (15%) of dynamic memory, leaving 54 bytes for local variables. Maximum is 64 bytes.
avrdude: AVR Part "attiny13a" not found.

Valid parts are:
  uc3a0512 = AT32UC3A0512
  c128     = AT90CAN128
  c32      = AT90CAN32
  c64      = AT90CAN64
...
 m8u2     = ATmega8U2
  t10      = ATtiny10
  t11      = ATtiny11
  t12      = ATtiny12
  t13      = ATtiny13
  t15      = ATtiny15
  t1604    = ATtiny1604
...

sleemanj, 2020-12-02:
Hi, which Programmer have you selected in the Tools > Programmer ?

caiopoit, 2020-12-03:
Arduino as ISP, and DIY ATTiny Arduino as ISP.

sleemanj, 2020-12-03:
Looks like avrdude doesn't know about the tiny13a 

The tiny13 and tiny13a however have the same signature bytes, so selecting Tiny13 instead of 13A should work, that your avrdude is complaining about signature bytes in that case may indicate another fail.  What is the exact signature output when you choose 13 instead of 13a?

caiopoit, 2020-12-03:
Attiny13 selected...

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\KylePolt\AppData\Local\Arduino15\packages\diy_attiny\hardware\avr\2020.10.10/avrdude.conf"

         Using Port                    : COM17
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding

caiopoit, 2020-12-03:
Whaaat, I just kept trying and it worked with the Attiny13 selected... The 13a still doesn't work... If they are the same then you may remove the option right?

sleemanj, 2020-12-03:
Unless you need to disable brown-out at runtime, they are the same.

It should be fixed probably by me making a custom avrdude.conf to include the attiny13a properly :+1:

caiopoit, 2020-12-03:
Alright, looking foward, your core is awesome! The first one not to brick my chips when burning bootloader.

utak3r, 2021-01-08:
Same using USBasp, switching to ATtiny13 helps. Also, note: when using USBasp on a fresh clean uC, often requires switching USBasp to a SLOW MODE. Once fuses are properly set, works normally :)

sleemanj, 2021-01-09:
>  often requires switching USBasp to a SLOW MODE

The usbasp firmware by @bperrybap et al includes an automatic speed reduction (and still allows the manual speed reduction) which should probably solve that for you :-)

I've been using the 1.06a release for years with no issues

https://github.com/bperrybap/usbasp/tree/1.06-alpha/bin/firmware/usbasp-v1.06-alpha-2016-05-18

helps to have a couple of USBAsp so you can update one with the other though ;-)

bperrybap, 2021-01-09:
The directory just above the one with the usbasp pre-built  images in my repository has the scripts to do the burning, including being able to use drag and drop to drop the file on top of the appropriate script  on Windows, Linux, & mac
https://github.com/bperrybap/usbasp/tree/1.06-alpha/bin/firmware

vks007, 2022-01-30:
I have the exact problem, I cant use AT13 as I am disabling brownout at run time. Is there a fix for this, I am using this library after many months and things seem to have changed. It all used to work earlier.

sleemanj, 2022-01-30:
If you are getting a signature error when trying to compile and upload for t13 (not t13a) then you have some other problem, sucjh as...

  1. you changed fuses to disable reset or spi programming
  2. you changed fuses to set clock frequency lower than your programmer is trying to talk to it
  3. you changed fuses to use an external clock and are not actually providing one
  4. you have a wiring issue, such as things connected to the MISO/MOSI or SCK pins interferring with the programming

As for the t13a problem, the Arduino distribution of avrdude.conf still does not include t13a.  You could locate the avrdude.conf your setup is using (look int he console window for the programming command and find which conf file it is using) and try adding a t13a section to it, such as https://github.com/MCUdude/MicroCore/blob/master/avr/avrdude.conf#L1346

vks007, 2022-01-30:
Thanks. let me try adding the config. 
But before that can you confirm that as per the pin diag you have the following are the only connections I have to do:
AT13a========Arduino
power=====5V
GND======GND
Pin7======13(SCK)
Pin6======12(MISO)
Pin5======11(MOSI)
Pin1======10(RESET)

sleemanj, 2022-01-30:
The connections depend on your programmer.  Sounds like you are using an Arduino as a programmer, which is not something I would recommend myself, but your connectiosn sound about right for that.

vks007, 2022-01-30:
Thanks, I was able to put in the avrdude config and upload the program to 13a. Dont know what was the issue in uploading earlier, must have been a connection as you mentioned.

closed 2023-04-19 by sleemanj