sleemanj/optiboot #22: digitalPinToInterrupt(PIN_X) does not work

skaman82, 2021-01-21:
Code:  
attachInterrupt( digitalPinToInterrupt(2) , isr, CHANGE ) ;

Compiler Error:
'sketch/IrNec.cpp: In function 'void nsIrNec::begin(uint8_t)':
IrNec.cpp:115:43: error: 'digitalPinToInterrupt' was not declared in this scope
   attachInterrupt( digitalPinToInterrupt(2) , isr, CHANGE ) ;
                                           ^
exit status 1
'digitalPinToInterrupt' was not declared in this scope'

Compiling for ATtiny85

sleemanj, 2021-01-21:
Timy 85 only has one external interrupt, just use it directly.

attachInterrupt( 0, isr, CHANGE ) ;

---- On Thu, 21 Jan 2021 23:38:44 +1300 Albert Kravcov <notifications@github.com> wrote ----

Code:
 attachInterrupt( digitalPinToInterrupt(2) , isr, CHANGE ) ;
Compiler Error:
 'sketch/IrNec.cpp: In function 'void nsIrNec::begin(uint8_t)':
 IrNec.cpp:115:43: error: 'digitalPinToInterrupt' was not declared in this scope
 attachInterrupt( digitalPinToInterrupt(2) , isr, CHANGE ) ;
 ^
 exit status 1
 'digitalPinToInterrupt' was not declared in this scope'
Compiling for ATtiny85

skaman82, 2021-01-30:
thanks, that helped

sleemanj, 2021-01-31:
I'll leave this open since I should fix it sometime.

sleemanj, 2022-03-12:
Fixed in 2022.3.10 release

closed 2022-03-12 by sleemanj