sleemanj/optiboot #23: Wrong readings from PulseIn Function

ToughAce, 2021-01-31:
Many thanks for providing this Attiny core!
I'm experiencing an issue with this very simple code:

`int unsigned long pwm = 0;

void setup() {
  pinMode(3, INPUT);
  pinMode(0, OUTPUT);
}

void loop() {
  pwm = pulseIn(3, HIGH);
  if (pwm > 1800) digitalWrite(0, HIGH);  
    else digitalWrite(0, LOW);
}`

It's supposed to be a simple RC switch and reads the PWM input from a RC receiver channel on Pin 3 and sets Pin 0 based on this.

This code works perfectly fine on an Attiny 45 with the dammelis Attiny core. But I would like to run on an Attiny 13 using this core and it turns out the PWM reading is by factor 1.5 too high. I have used a servo tester to generate pulses between 800 and 2200us and sent the readings to a serial monitor and they are between 1200 and 3300us. Same happen when I use the same Attiny 45 with this core instead of the dammelis core. I have used the presettings and burned the fuses to the chip. I have also used several other fuse combinations without success.
For now I've adjusted the if statement to use 2700 instead of 1800 and it switches as inteded when the input goes over 1800us.
Nevertheless, would be interested to know what I'm doing wrong

Thanks!

sleemanj, 2021-01-31:
Mmm.  Probably not doing anything wrong, could just be a bug in my code :)

You could try changing the "Millis Accuracy" setting (Tools Menu) and see if that makes a difference.  You could also try [SpenceKonde's original ATTInyCore](https://github.com/spencekonde/ATTinyCore) with a Tiny45 and see if that produces a better result there, Spence's doesn't support Tiny13, but it could narrow it down to something in my changes.

sleemanj, 2021-01-31:
Actually, now I dig into it, yes, this all came up before :)

https://github.com/SpenceKonde/ATTinyCore/issues/384

SpenceKonde's original ATTinyCore has a fix but I have not rolled it in yet.

sleemanj, 2022-03-12:
The 2022.3.10 release should incorporate the changes.  Hopefully they work for the T13 (not tested)

closed 2022-03-12 by sleemanj