sleemanj/optiboot #24: Can't compile anymore an mac os 11.5 or ATTiny10

skaman82, 2021-08-28:
Since I have upgraded to mac os 11.5 BigSur Arduino is not longer compiling code (in my case) for ATTINY10. 
Following error shows up: "fork/exec /Users/myusername/Library/Arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/avr-g++: bad CPU type in executable"

sleemanj, 2021-08-29:
https://forum.arduino.cc/t/avr-g-bad-cpu-type-in-executable/595790/2

sleemanj, 2021-08-29:
If you can compile for the Arduino Uno and it's just the ATTiny10 (and others in my package) which are broken, then you could uninstall my package (from Boards Manager) and then do a Manual Install ( https://github.com/sleemanj/optiboot/blob/master/dists/README.md#manual-installation-not-recommended ) this might work for you.  If you can not compile for Arduino Uno either, with the same error, then that would mean that the Arduino ecosystem still doesn't work for your type of Mac OS as in the link above.

Sorry I don't have a Mac so I can't test on one.

skaman82, 2021-08-29:
I can compile or other targets. The problem only occurs with the targets in your package. I will try the manual installation and get back to you.

skaman82, 2021-08-30:
@sleemanj I did the manual installation but now I am getting following error when trying to compile:

/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find crtattiny10.o: No such file or directory
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: skipping incompatible /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/libm.a when searching for -lm
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lm
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: skipping incompatible /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/libgcc.a when searching for -lgcc
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lgcc
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: skipping incompatible /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/libm.a when searching for -lm
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lm
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: skipping incompatible /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/libc.a when searching for -lc
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lc
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lattiny10
collect2: error: ld returned 1 exit status
exit status 1

skaman82, 2021-08-30:
I get the error only if I choose AT4, AT5, AT9 and AT10.
AT13, AT24, AT44, AT84, AT25, AT45 and AT85  seems to compile normally now.

sleemanj, 2021-08-31:
I suspected that might be the issue, it is this bug in the Arduino toolchain: https://github.com/arduino/toolchain-avr/issues/65

Unfortunately I can't fix it, up to Arduino.

sleemanj, 2022-03-12:
This won't help you on the 64bit Mac, but if anybody else comes along, make sure to use the "-gcc5.4" of this core for 4/5/9/10 support.  The -gcc7.3 version will exhibit thi problem (on all platforms not just Mac).

The 64bit Macs can (I think) only use the -gcc7.3 version so they always have the problem.   Unfortunately until that ticket https://github.com/arduino/toolchain-avr/issues/65 is sorted out in the Arduino Toolchain, I can't do anything about it for Macs.