sleemanj/optiboot #27: millis does not work on attiny 13

noName2005, 2022-03-09:
I'm using an attiny13a but when I select that option I get an error saying it can't find the board. when I program with attiny13 selected it works just fine. I'm also using a library to run a stepper motor with the attiny. Previously everything worked fine. I was able to run the stepper motor while also using millis. Im now working on creating one-wire communication between Arduino and the attiny13, but I noticed the program stops whenever millis is mentioned. 

`#include "iotool_attiny13a_lowlevel.h"
#define STEPPER_SLEEP _delay_us(2000)
#define address 2
int steps;
int dig1;
int dig2;
int dig3;
int dig4;
int fin;
int num;

//Millis
unsigned long start;
unsigned long endI;
unsigned long waitAddress = 6000;
unsigned long waitDigit = 18000;
int initialize = 4000;

int main()
{
  // --- setup ---
  PINMODE_PB0_OUTPUT; DIGITALWRITE_PB0_LOW;
  PINMODE_PB1_OUTPUT; DIGITALWRITE_PB1_LOW;
  PINMODE_PB2_OUTPUT; DIGITALWRITE_PB2_LOW;
  PINMODE_PB4_OUTPUT; DIGITALWRITE_PB4_LOW;
  PINMODE_PB3_INPUT;
  //  PINMODE_PB4_OUTPUT; DIGITALWRITE_PB4_LOW;
  ADC_DISABLE;

  // --- loop ---
  while (1)
  {
    loop();
  }
}

void stepperOff() {
  DIGITALWRITE_PB0_LOW;
  DIGITALWRITE_PB1_LOW;
  DIGITALWRITE_PB2_LOW;
  DIGITALWRITE_PB4_LOW;
}

void stepper1() {
  DIGITALWRITE_PB4_HIGH;
  STEPPER_SLEEP;
  DIGITALWRITE_PB4_LOW;
}

void stepper2() {
  DIGITALWRITE_PB4_HIGH;
  DIGITALWRITE_PB2_HIGH;
  STEPPER_SLEEP;
  DIGITALWRITE_PB4_LOW;
  DIGITALWRITE_PB2_LOW;
}

void stepper3() {
  DIGITALWRITE_PB2_HIGH;
  STEPPER_SLEEP;
  DIGITALWRITE_PB2_LOW;
}

void stepper4() {
  DIGITALWRITE_PB1_HIGH;
  DIGITALWRITE_PB2_HIGH;
  STEPPER_SLEEP;
  DIGITALWRITE_PB1_LOW;
  DIGITALWRITE_PB2_LOW;
}

void stepper5() {
  DIGITALWRITE_PB1_HIGH;
  STEPPER_SLEEP;
  DIGITALWRITE_PB1_LOW;
}

void stepper6() {
  DIGITALWRITE_PB0_HIGH;
  DIGITALWRITE_PB1_HIGH;
  STEPPER_SLEEP;
  DIGITALWRITE_PB0_LOW;
  DIGITALWRITE_PB1_LOW;
}

void stepper7() {
  DIGITALWRITE_PB0_HIGH;
  STEPPER_SLEEP;
  DIGITALWRITE_PB0_LOW;
}

void stepper8() {
  DIGITALWRITE_PB4_HIGH;
  DIGITALWRITE_PB0_HIGH;
  STEPPER_SLEEP;
  DIGITALWRITE_PB4_LOW;
  DIGITALWRITE_PB0_LOW;
}

//functions to read state of wire and count
//int Read(int wait) {
//  int state;
//  int Lstate=0;
//  for(start=millis(); (millis()-start)<wait; ){
//    if(DIGITALREAD_PB3_HIGH){
//      state=1;
//      DIGITALWRITE_PB0_HIGH;
//    }
//    if(DIGITALREAD_PB3_LOW){
//      state=0;
//      DIGITALWRITE_PB0_LOW;
//    }
//    if(state!=Lstate){
//      if(DIGITALREAD_PB3_HIGH){
//        num++;
//      }else{}
//    }
//    Lstate=state;
//  }
//  return num;
//}

//functions to read state of wire and count
//int Read(int wait){
//  start=millis();
//  int now;
//  while(now-start<wait){
//    while(DIGITALREAD_PB3_LOW&&now-start<wait){
//      DIGITALWRITE_PB0_LOW;
//      now=millis();
//    }
//    while(DIGITALREAD_PB3_HIGH&&now-start<wait){
//      DIGITALWRITE_PB0_HIGH;
//      now=millis();
//    }
//    num++;
//    now=millis();
//  }
//  return num;
//}

void loop() {
  //test program to test millis
  DIGITALWRITE_PB1_HIGH;
  delay(500);
  DIGITALWRITE_PB1_LOW;
  delay(500);
  //dig1=Read(waitDigit);
  //for(start=millis(); (millis()-start)<4000; ){}   //delay using millis (tried but failed)
  start = millis();
  while ((start + 4000) > millis()) {}    //other type of delay using millis (also failed)
  DIGITALWRITE_PB1_HIGH;
  delay(500);
  DIGITALWRITE_PB1_LOW;
  delay(500);
  //  for(int i=0; i<dig1; i++){
  //      DIGITALWRITE_PB1_HIGH;
  //      delay(1000);
  //      DIGITALWRITE_PB1_LOW;
  //      delay(1000);
  //  }
  delay(60000);
}`

When power is applied the LED connected to PB1 goes high 500ms then low 500ms, then it just stops instead of flashing again.
As I previously said, millis has worked before with the stepper library. I think one of the newer updates to the DIY attiny programmer has done something weird or my programming is just trash and I missed something obvious.

library for stepper motor: https://github.com/iotool/IOTool_ATtiny13a_LowLevel

sleemanj, 2022-03-09:
You are implementing your own main() therefore the millis timer is never established.Message ID: ***@***.***>

closed 2022-03-09 by sleemanj

noName2005, 2022-03-11:
Is there a way to only establish millis timer? With an external library or something? I can not use void setup as the file sketch size gets too big.

sleemanj, 2022-03-11:
No.  Use the menu options to determine what things to initialise to reduce size. ---- On Sat, 12 Mar 2022 03:30:58 +1300  ***@***.***  wrote ----
Is there a way to only establish millis timer? With an external library or something? I can not use void setup as the file sketch size gets too big.