sleemanj/optiboot #30: is map() implemented to work for attiny13?

darkmetalman, 2024-07-04:
Hello
Is map() implemented to work for attiny13?
My code is not working with when I use the map() arduino function.

It works when I flash using MCUdude's MicroCore.

Regards!

sleemanj, 2024-07-04:
Hmmm.

Are you passing in constant values for the ranges, or are they variables?

sleemanj, 2024-07-04:
If you are using variable for the start/end ranges (eg `map(x,y,z,j,h)`), there is a bug which would prevent it from compiling, put this in your sketch to correct it

    uint32_t _map( uint32_t x, uint32_t in_min, uint32_t in_max, uint32_t out_min, uint32_t out_max)
    {
      return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
    }

if you are using constants for the ranges (eg `map(x, 1,1023, 1,255)`) then the bug would not present itself.

darkmetalman, 2024-07-04:
**Thanks!!! that solved the problem.**

closed 2024-07-04 by darkmetalman